google-coral/edgetpu-platforms #1: Basic example failing with RuntimeErrror: Failed to allocate tensors.

stonehippo, 2019-04-23:
I tried running the latest beta release (1.9.2) on a Pi Zero W with the first run demo in the documentation. The following error occurred:

```
Traceback (most recent call last):
  File "classify_image.py", line 67, in <module>
    main()
  File "classify_image.py", line 58, in main
    engine = ClassificationEngine(args.model)
  File "/usr/local/lib/python3.5/dist-packages/edgetpu/classification/engine.py", line 38, in __init__
    super().__init__(model_path)
  File "/usr/local/lib/python3.5/dist-packages/edgetpu/swig/edgetpu_cpp_wrapper.py", line 308, in __init__
    this = _edgetpu_cpp_wrapper.new_BasicEngine(*args)
RuntimeError: Failed to allocate tensors.
```
I know this release is not official, but any idea what might be going on and how to resolve it? I’ve run the test scripts on a Pi 3 B+ using the same USB accelerator successfully, so I know I’ve got a working unit.

mtyka, 2019-04-26:
Did you try un-plugging / replugging the accelerator and/or rebooting ? I found that right after installation you have to replug the accelerator, otherwise this error occurs.

mtyka, 2019-04-26:
Just reinstalled with 1.9.2 on a clean pi zero w and the ccanned example (https://coral.withgoogle.com/docs/accelerator/get-started/)  worked great.

I'd suggest a clean reinstall I guess.

closed 2019-04-26 by mtyka

stonehippo, 2019-04-26:
Yeah, I tried unplug/plug in, since the udev rules change on install. 

I'll do clean install of Raspbian and 1.9.2, and also try it one another Pi Zero W.

stonehippo, 2019-04-26:
@mtyka question: how did you connect the accelerator to the Pi Zero W? Directly with a USB-C to micro cable? Or via something like a powered USB hub? I'm wondering if my issue is power, not software.

stonehippo, 2019-04-27:
Ugh, I think this was because I wasn’t using a USG OTG adapter. I’ll give that a try and confirm, but I’m pretty sure that’s it.

PeterMalkin, 2019-04-27:
Good point about power. I'd recommend going through a powered hub, since Edge TPU can draw up to 2W during 100% utilization. So if pi0 cant supply enough voltage, your accelerator may brown out.

stonehippo, 2019-04-27:
@mtyka (and anyone who comes across this one), I figured it out. I realized that on the Pi Zero in question, I had removed the default Raspbian `pi` user, and the user account I was using is only in the `sudo` group. When I tried the example code like this it worked:

```
$ sudo python3 classify_image.py \
--model ~/Downloads/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu. \
--label ~/Downloads/inat_bird_labels.txt \
--image ~/Downloads/parrot.jpg 
```
I checked the `udev` rules set up in the install script, and it applies to the `plugdev` group. So I needed to add my account to that group:

```
$ sudo usermod -aG plugdev [my username]
```

Problem solved.

Banus, 2019-05-10:
> Did you try un-plugging / replugging the accelerator and/or rebooting ?

This. I tried all the other solutions, but in my case the device started working only after rebooting the system.

mali30, 2019-06-27:
hello I have unplugged and replugged. shut down and restarted and nothing works. can anyone suggest any other ideas on why the board is not being recognized. I keep seeing UniCorp Chip instead of Google <>.

Ryan-Johnson-1315, 2020-02-14:
I am having the same problem still. Is there any update on this? This is my error: 

`
Image list successfully parsed! Category Num =  10
Traceback (most recent call last):
  File "imprinting_learning.py", line 245, in <module>
    main()
  File "imprinting_learning.py", line 199, in main
    shape = _get_required_shape(args.model_path)
  File "imprinting_learning.py", line 140, in _get_required_shape
    tmp = BasicEngine(model_path)
  File "/usr/lib/python3/dist-packages/edgetpu/basic/basic_engine.py", line 92, in __init__
    self._engine = BasicEnginePythonWrapper.CreateFromFile(model_path)
RuntimeError: Internal: :77 tf_lite_type != kTfLiteUInt8 (1 != 3)Node number 0 (edgetpu-custom-op) failed to prepare.
Failed to allocate tensors.

`