google-coral/edgetpu-platforms #4: USB Coral with pass-through into Ubuntu 19.04 in VirtualBox: Basic example fails with "Failed to allocate tensors"

ChristophDerndorfer, 2019-05-16:
I tried running the basic example with the USB Coral that's connected to an Ubuntu 19.04 guest in a VirtualBox on a Windows 10 host via USB passthrough.

# sudo lsusb shows that the dongle is connected successfully:
Bus 002 Device 004: ID 18d1;9302 Google Inc

However, when I run the example command (python3 classify_image.py \
--model ~/Downloads/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite \
--label ~/Downloads/inat_bird_labels.txt \
--image ~/Downloads/parrot.jpg) I only get:
Runtime error: Failed to allocate tensors.

A subsequent # sudo lsusb doesn't show the USB Coral device anymore.

During the installation I also ran into the issue mentioned [here](https://github.com/f0cal/google-coral/issues/13) and subsequently also executed the command that's supposed to enable the dirty work-around.

Looking at other issues I also tried # sudo usermod -aG plugdev [my username] and reboots (mentioned in #1 ) but to no avail.

IoTman, 2019-05-20:
Try connecting the Coral USB device to your PC via a USB hub, I've had similar issues with other USB3 devices that works with some USB hubs, and not others.  USB3 devices start up in USB2 mode, then a driver is loaded and the device reboots into USB3 mode.  My failing devices/hub would cause similar symptoms, ie devices would disappear from the lsusb, and need repowered to reset. This was just plain Ubuntu, without your extra VM fun.

ChristophDerndorfer, 2019-05-21:
Thanks for the recommendation. I had no luck with an unpowered (and fairly old) USB hub so I ordered a new powered one and will then report back.

ChristophDerndorfer, 2019-05-22:
Unfortunately using the new powered USB hub didn't change anything. It also doesn't matter whether I connect the USB hub via my USB 2.0 or USB 3.0 ports (using a Lenovo T430) and use VirtualBox's USB 2.0 or USB 3.0 controller. I'll try to continue the investigation on the weekend.

ChristophDerndorfer, 2019-05-22:
[These instructions](https://dev.to/kojikanao/coral-edgetpu-usb-with-virtualbox-57e1) by @koji finally provided me with a solution:

Aside of adding the "Global Unichip Corp [0100]" device to VirtualBox's USB devices you also need to add a second filter for an arbitrarily called USB device with Vendor ID: 18d1 and Product ID: 9302. Then the basic example suddenly worked like a charm:

[image]

closed 2019-05-22 by ChristophDerndorfer

hb020, 2019-06-15:
For the record: VMWare has the same problem. You need to pass through the "Google USB Device". If you see "Global Unichip Corp" in lsusb in the VM, or in the VMWare USB console, then it will not work. A reboot of the VM (not of the host) may be needed to see that device.
[image]

hb020, 2019-06-17:
Now I got this working once on my MBP, and now I am no longer able to get it working. Tried SMC reset, rebooting the host and the VM, VirtualBox with the above tweak of the extra USB device, VMWare Fusion with and without the equivalent tweak in the vmx file, all no luck. I no longer get the "18d1:9302 Google Inc" device. Anyone having a Macbook Pro that has succeeded in keeping it working?
So in essence, the solution above is partial, those on Mac might hit extra hurdles.