google-coral/edgetpu-platforms #6: Add Support Raspberry Pi 2

wgcv, 2019-05-29:
Hi
I was checking and the coral has support to RaspberryPi 2 Model B but in the installer `install.sh` doesn't have.

https://coral.withgoogle.com/docs/accelerator/get-started/ 
> And yes, this means Raspberry Pi is supported. However, we officially support only Raspberry Pi 2/3 Model B/B+ running Raspbian. Unofficially, support for Raspberry Pi Zero is also available (install the TAR from GitHub on your Pi Zero, instead of the one below).

I would add these lines to `install.sh`
```
  elif [[ "${MODEL}" == "Raspberry Pi 2 Model B Rev"* ]]; then
    info "Recognized as Raspberry Pi 2 B+."
    LIBEDGETPU_SUFFIX=arm32
    HOST_GNU_TYPE=arm-linux-gnueabihf
  fi
```

mtyka, 2019-06-02:
I don't have a Pi 2 to test this with - can you confirm that this worked on an actual Pi 2? If so I can add unofficial support to the repo here.

wgcv, 2019-06-02:
yes

mtyka, 2019-06-03:
Ok, added a release binary for Pi 2 - thanks !!

closed 2019-06-03 by mtyka