google-coral/edgetpu-platforms #8: Failed to allocate tensors continued..

mali30, 2019-06-27:
The board is not recognized and I keep getting the `failed to allocate tensors` error. I have restarted, shutdown, and also connected and disconnected the usb. No avail. I have also added it to the plugdev group by running the command

`sudo usermod -aG plugdev root`

Can someone please give any other suggestion on how I can overcome this issue?

mtyka, 2019-07-08:
What system/os versions are you using ? Rpi0 or 3 ? What code are you running ?

mali30, 2019-07-08:
I figured out a quick fix. Ran the get started on the google coral board page and it works after. The problem was it was not detecting the Coral Board. It kept saying Unicorp Chip instead of Google Inc.

mtyka, 2019-07-09:
Yes, it does say Unicorp Chip. But that's the TPU - great i'm gald you got it working !

closed 2019-07-09 by mtyka