google-coral/edgetpu-platforms #9: Install script does not install whl file

mtuxpe, 2019-07-03:
My target hardware is raspberry pi zero w.
I am using 1.92 release and pi zero w was detected.
Python API does not install.
I download from https://github.com/google-coral/edgetpu-platforms/releases/download/v1.9.2/edgetpu_api_1.9.2.tar.gz

# Python API.
WHEEL=$(ls ${SCRIPT_DIR}/edgetpu-*-py3-none-any.whl 2>/dev/null)
if [[ $? == 0 ]]; then
  info "Installing Edge TPU Python API..."
  sudo python3 -m pip3 install --no-deps "${WHEEL}"
  info "Done."
fi

mtyka, 2019-07-11:
I can't reproduce this, what sort of error are you getting ?

mtuxpe, 2019-07-12:
> I can't reproduce this, what sort of error are you getting ?

No error  message.
The message info "Installing Edge TPU Python API..." is not printed on console

mtyka, 2019-07-26:
I need more information to help debug this, since the script works on my system. 

In the meantime you can also use this prebuilt image for raspberry pi 0:
https://github.com/google-coral/edgetpu-platforms/releases/download/v1.9.2/rpi0-raspbian-stretch-edgetpu-1.9.2.img.gz

closed 2019-07-26 by mtyka