google-coral/edgetpu-platforms #10: Fails to load `edgetpu.swig._edgetpu_cpp_wrapper` on Raspbian Buster/Python 3.7

stonehippo, 2019-07-07:
I recently moved to the latest Raspbian Buster release, which includes Python 3.7. After reinstalling the Pi Zero W version of the EdgeTPU support, I’m getting the following error when attempting to run the `classify_image.py` demo:

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/edgetpu/swig/edgetpu_cpp_wrapper.py", line 18, in swig_import_helper
    return importlib.import_module(mname)
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 965, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'edgetpu.swig._edgetpu_cpp_wrapper'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "classify_image.py", line 19, in <module>
    from edgetpu.classification.engine import ClassificationEngine
  File "/usr/local/lib/python3.7/dist-packages/edgetpu/classification/engine.py", line 17, in <module>
    from edgetpu.basic.basic_engine import BasicEngine
  File "/usr/local/lib/python3.7/dist-packages/edgetpu/basic/basic_engine.py", line 15, in <module>
    from edgetpu.swig.edgetpu_cpp_wrapper import BasicEngine
  File "/usr/local/lib/python3.7/dist-packages/edgetpu/swig/edgetpu_cpp_wrapper.py", line 21, in <module>
    _edgetpu_cpp_wrapper = swig_import_helper()
  File "/usr/local/lib/python3.7/dist-packages/edgetpu/swig/edgetpu_cpp_wrapper.py", line 20, in swig_import_helper
    return importlib.import_module('_edgetpu_cpp_wrapper')
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ModuleNotFoundError: No module named '_edgetpu_cpp_wrapper'
```

I’m not really sure about swig and using C/C++ code with Python yet, but aliasing `/usr/local/lib/python3.7/dist-packages/edgetpu/swig/edgetpu_cpp_wrapper.py` to `/usr/local/lib/python3.7/diet-packages/edgetpu/_edgetpu_cpp_wrapper.py` the module resolution worked. However, this led to another error:

```
Traceback (most recent call last):
  File "classify_image.py", line 19, in <module>
    from edgetpu.classification.engine import ClassificationEngine
  File "/usr/local/lib/python3.7/dist-packages/edgetpu/classification/engine.py", line 17, in <module>
    from edgetpu.basic.basic_engine import BasicEngine
  File "/usr/local/lib/python3.7/dist-packages/edgetpu/basic/basic_engine.py", line 15, in <module>
    from edgetpu.swig.edgetpu_cpp_wrapper import BasicEngine
  File "/usr/local/lib/python3.7/dist-packages/edgetpu/swig/edgetpu_cpp_wrapper.py", line 21, in <module>
    _edgetpu_cpp_wrapper = swig_import_helper()
  File "/usr/local/lib/python3.7/dist-packages/edgetpu/swig/edgetpu_cpp_wrapper.py", line 18, in swig_import_helper
    return importlib.import_module(mname)
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "/usr/local/lib/python3.7/dist-packages/edgetpu/swig/_edgetpu_cpp_wrapper.py", line 101, in <module>
    class SwigPyIterator(_object):
  File "/usr/local/lib/python3.7/dist-packages/edgetpu/swig/_edgetpu_cpp_wrapper.py", line 110, in SwigPyIterator
    __swig_destroy__ = _edgetpu_cpp_wrapper.delete_SwigPyIterator
AttributeError: module 'edgetpu.swig._edgetpu_cpp_wrapper' has no attribute 'delete_SwigPyIterator'
```

Based on this, I presume that the swig binding in Raspbian Buster are broken somehow. 

I’m going to try a clean install of Buster with 1.9.2 (rather than the upgrade I’m using now) to see if that makes a difference.

MrJBSwe, 2019-07-07:
This is my fix on Nividia Jetson Nano and Nanopi M4 ( might be useful on Pi4 )

cd /usr/local/lib/python3.6/dist-packages/edgetpu/swig/
sudo cp _edgetpu_cpp_wrapper.cpython-35m-aarch64-linux-gnu.so edgetpu_cpp_wrapper.so

stonehippo, 2019-07-07:
Ok, that got me a bit further, but now it's complaining about other missing libs. Pretty sure it needs to be compiled for Python 3.7.

MrJBSwe, 2019-07-08:
Uninstall 3.7 and use 3.6 ?

mtyka, 2019-07-08:
Which device are you using - Rpi Zero or Rpi3 ? 

I just reproduced your error using a clean Raspbian Buster and a Rpi3. A workaround was (as suggested above) to do:
cd /usr/local/lib/python3.7/dist-packages/edgetpu/swig/
sudo cp _edgetpu_cpp_wrapper.cpython-35m-aarch64-linux-gnu.so _edgetpu_cpp_wrapper.cpython-37m-aarch64-linux-gnu.so 

This worked for me and I can now run the examples here:
https://coral.withgoogle.com/docs/accelerator/get-started/

stonehippo, 2019-07-08:
@mtyka I'm trying this on a Pi Zero, so I can try the same with the armhf .so.

mtyka, 2019-07-09:
Ah yes, I meant the armhf.so not the aarch64 one - sorry for the typo.
LMK if it works - in theory it should work since the libedgetpu.so is a seperate lib that's independent from the python version.

stonehippo, 2019-07-10:
@mtyka I tried the cp on my upgraded instance of Buster; no dice. There was an import error for `libpython3.5m.so.1.0`, which implies that maybe the armhf wrapper is dynamically linked. I was able to fix that by explicitly installed `libpython35-dev`, but that lead to another missing dependency (`libgflags`) and now it's whack-a-mole.

I then tried a clean install of Raspian Buster Lite, and only installed apt updates, and the edgetpu API. Still no dice; same issues as with the upgraded Raspbian. Looks like there needs to be a recompilation to support Raspbian Buster on RPi Zero and Python 3.7. :(

mtyka, 2019-07-10:
I'll try and take a look this week. For now I recommend to stick with Raspbian Stretch on Pi 0

mtyka, 2019-07-26:
I haven't been able to make this work unfortunately. For now please stick with stretch.
For convenience i've added a prebuilt image for rpi0:
https://github.com/google-coral/edgetpu-platforms/releases/download/v1.9.2/rpi0-raspbian-stretch-edgetpu-1.9.2.img.gz

closed 2019-07-26 by mtyka

LifelongLearnerOki, 2019-12-29:
@mtyka Currently, I'm working on a PiCar-V with Respberry 3+, but I have the same issue with "No module named 'edgetpu'.

Before, I tried following 

`wget https://dl.google.com/coral/edgetpu_api/edgetpu_api_latest.tar.gz -O edgetpu_api.tar.gz --trust-server-names
tar xzf edgetpu_api.tar.gz
cd edgetpu_api
bash ./install.sh`

and I always get this error:

`Traceback (most recent call last):
  File "code/coco_object_detection.py", line 9, in <module>
    import edgetpu.detection.engine
ModuleNotFoundError: No module named 'edgetpu'`

I was trying to follow your advice, but couldn't make it. Would be perfect if you could help me to handle this issue:-)

cd /usr/local/lib/python3.7/dist-packages/edgetpu/swig/
sudo cp _edgetpu_cpp_wrapper.cpython-35m-arm-linux-gnueabihf.so _edgetpu_cpp_wrapper.cpython-37m-arm-linux-gnueabihf.so