google-coral/edgetpu-platforms #12: Illegal Instruction

JVTEAM, 2019-07-29:
I've had this happen on 2 different scripts.  coral usb is working on the non-live video functions, but I get this when trying to run live on the raspberry pi zero.

python3 detect_video.py --model mobilenet_ssd_v2/mobilenet_ssd_v2_coco_quant_postprocess_edgetpu.tflite --labels mobilenet_ssd_v2/coco_labels.txt
[INFO] parsing class labels...
[INFO] loading Coral model...
W /home/pi/edgetpu-ml-cpp-release-rpi0-1.9.2/darwinn/third_party/darwinn/driver/package_registry.cc:65] Minimum runtime version required by package (5) is lower than expected (10).
[INFO] starting video stream...
Illegal instruction

mtyka, 2019-07-30:
Where did detect_video.py come from ? Can you link to the source ?

JVTEAM, 2019-07-30:
This github page....

https://github.com/google-coral/examples-camera/tree/master/opencv

JVTEAM, 2019-07-30:
https://www.pyimagesearch.com/2019/05/13/object-detection-and-image-classification-with-google-coral-usb-accelerator/

mtyka, 2019-07-31:
"Illegal instruction" suggests you're somehow running binary code for armv7 or 8 on your pi zero which is armv6. Given that you're only seeing this when using a video stream, my guess is that this is coming from open cv - how did you install that on your pi zero ? Was is the right package for armv6 or did you compile it on the pi zero ?
Maybe instead, try the raspicam example 
https://github.com/google-coral/examples-camera/tree/master/raspicam
That one runs the classification model with a camera stream and works fine on a pi zero.
It would be easy to adapt to the detection model. I can confirm that mobilenet_ssd_v2_face_quant_postprocess_edgetpu.tflite runs totally fine on a pi zero by itself.

closed 2019-07-31 by mtyka