google-coral/edgetpu-platforms #15: AMD E-450 support

jalbstmeijer, 2019-08-13:
Hi,

Was expecting the AMD E-450 just to work with  default x86_64 support from https://dl.google.com/coral/edgetpu_api/edgetpu_api_latest.tar.gz.
But I get a segfault while running the parrot demo.

[ 3198.837222] traps: python3[1623] trap invalid opcode ip:7fa88b4a380e sp:7ffdb12eec60 error:0 in _edgetpu_cpp_wrapper.cpython-36m-x86_64-linux-gnu.so[7fa88b46f000+229000]

Googling for similar issues, it seems that such issue could be caused by
compiles utilizing compile options which are not supported by the target
CPU.

Should the AMD E-450 be considered as a different platform?

I attached coredump collected this way:

coredumpctl -o core_dump dump /usr/bin/python3.6 > core_dump.txt
gunzip core_dump

[core_dump.gz](https://github.com/google-coral/edgetpu-platforms/files/3496149/core_dump.gz)
[core_dump.txt](https://github.com/google-coral/edgetpu-platforms/files/3496150/core_dump.txt)

$ lscpu
Architecture:        x86_64
CPU op-mode(s):      32-bit, 64-bit
Byte Order:          Little Endian
Address sizes:       36 bits physical, 48 bits virtual
CPU(s):              2
On-line CPU(s) list: 0,1
Thread(s) per core:  1
Core(s) per socket:  2
Socket(s):           1
NUMA node(s):        1
Vendor ID:           AuthenticAMD
CPU family:          20
Model:               2
Model name:          AMD E-450 APU with Radeon(tm) HD Graphics
Stepping:            0
CPU MHz:             900.690
CPU max MHz:         1650.0000
CPU min MHz:         825.0000
BogoMIPS:            3299.87
Virtualization:      AMD-V
L1d cache:           32K
L1i cache:           32K
L2 cache:            512K
NUMA node0 CPU(s):   0,1
Flags:               fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_good nopl nonstop_tsc cpuid extd_apicid aperfmperf pni monitor ssse3 cx16 popcnt lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch ibs skinit wdt hw_pstate vmmcall arat npt lbrv svm_lock nrip_save pausefilter

Gr, J

mtyka, 2019-08-14:
Yes I expect AMD E-450 would need to be custom compiled. Unfortunately we're not currently able to support this platform.

closed 2019-08-14 by mtyka

jalbstmeijer, 2019-08-14:
Thanks for the clear reply,
Just asking, any chance the code will be released any time soon so we can compile it our selves?

PeterMalkin, 2019-08-14:
We are looking at opensourcing the SWIG part of the wrapper. If you rebuild that, there's a chance it may run fine. The next release should probably have it, but dont have exact date yet.