google-coral/edgetpu-platforms #16: ImportError: No module named '_edgetpu_cpp_wrapper' on Python 3.5

ssepulveda, 2019-08-20:
I'm trying to run Coral USB accelerator on a STM32MP1-DK2 (this is an ARM32v7 device). I'm following script from https://github.com/google-coral/edgetpu-platforms/releases/tag/v1.9.2 for a RPi Zero and got the following error when trying to run classify_image.py demo:

```
Traceback (most recent call last):
  File "/usr/lib/python3.5/site-packages/edgetpu/swig/edgetpu_cpp_wrapper.py", line 18, in swig_importr
    return importlib.import_module(mname)
  File "/usr/lib/python3.5/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 985, in _gcd_import
  File "<frozen importlib._bootstrap>", line 968, in _find_and_load
  File "<frozen importlib._bootstrap>", line 957, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 666, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 577, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 938, in create_module
  File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
ImportError: /usr/lib/python3.5/site-packages/edgetpu/swig/_edgetpu_cpp_wrapper.cpython-35m-arm-linux-_

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "classify_image.py", line 19, in <module>
    from edgetpu.classification.engine import ClassificationEngine
  File "/usr/lib/python3.5/site-packages/edgetpu/classification/engine.py", line 17, in <module>
    from edgetpu.basic.basic_engine import BasicEngine
  File "/usr/lib/python3.5/site-packages/edgetpu/basic/basic_engine.py", line 15, in <module>
    from edgetpu.swig.edgetpu_cpp_wrapper import BasicEngine
  File "/usr/lib/python3.5/site-packages/edgetpu/swig/edgetpu_cpp_wrapper.py", line 21, in <module>
    _edgetpu_cpp_wrapper = swig_import_helper()
  File "/usr/lib/python3.5/site-packages/edgetpu/swig/edgetpu_cpp_wrapper.py", line 20, in swig_importr
    return importlib.import_module('_edgetpu_cpp_wrapper')
  File "/usr/lib/python3.5/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: No module named '_edgetpu_cpp_wrapper'
```

Any ideas ?

mtyka, 2019-08-21:
Hmm, i'm not sure, i've never tried it on a STM32MP1-DK2. Possible you may have to rename the .so files  in /usr/lib/python3.5/site-packages/edgetpu/swig/ on that platform.

closed 2019-08-21 by mtyka

smertrios, 2019-09-08:
I'm having the same issue on a non-raspi distro / device.

=> undefined symbol: _ZN6google14FlagRegistererC1EPKcS2_S2_S2_PvS3_

module is being found & loaded just fine but it fails to run... looks like a dependency is not met.

could we get a list of dependencies when installing on a custom distro? thanks.

powderluv, 2019-09-22:
I had python3.7 (built mendel from master branch) so I did:

cd /usr/lib/python3/dist-packages/edgetpu/swig/
sudo ln -s _edgetpu_cpp_wrapper.cpython-35m-aarch64-linux-gnu.so _edgetpu_cpp_wrapper.cpython-37m-aarch64-linux-gnu.so