google-coral/edgetpu-platforms #19: Add support macOS

chertov, 2019-09-10:
Hi! It will be great to develop and to use coral usb on macOS platform!
Do you have any plans to support macOS?

alsuren, 2019-09-10:
Hi @chertov I managed to get the coral usb accelerator working under Vagrant on macOS. Please try out the instructions at https://github.com/google-coral/project-posenet/pull/3 and tell me how you get on.

I ended up running my models on the CPU, so I haven't really touched it since I made the PR. If I get time over the next couple of weeks, I will try to find a way to generalise it so that it can be added to this repo as well.

powderluv, 2019-10-29:
@Namburger we will need a libedgetpu.so build for OSX to be able to use the USB accelerator on Macs. Any chance you can provide one ?

Namburger, 2019-10-29:
@powderluv the only 2 .so file we have right now is for the arm && x86_64 right now. I believe the Mac cpu should be able to use x86_64 right?

powderluv, 2019-10-29:
no you will an x86_64 .so built on OSX since the formats are different.

Namburger, 2019-10-29:
@powderluv sorry I got confused with the compiler (I was still at home), does the k8 build from [here](https://github.com/google-coral/edgetpu/tree/master/libedgetpu/throttled) works with MacOS?

Namburger, 2019-10-29:
@powderluv 
Update: We are working on a MacOS port now, it is just not available at this moment.

chertov, 2020-01-02:
Any news about native macos support?

rhamnett, 2020-01-24:
Also eagerly awaiting this

AVanVlack, 2020-01-30:
I am seeing a getting started guide for mac on the website (https://coral.ai/docs/accelerator/get-started/#on-mac) but the link to the needed runtime is dead. Anyone have the correct link? Looks like the version was posted yesterday.

iPsych, 2020-01-30:
@AVanVlack 
Fantastic update! 
But, it seems they did not uploaded actual file yet.

Namburger, 2020-01-30:
Hi guys, edgetpu supports for macos and windows are now available [here](https://coral.ai/docs/accelerator/get-started/#on-mac)

iPsych, 2020-01-30:
@Namburger 
Sadly, the link is still dead.
https://dl.google.com/coral/edgetpu_runtime_20200128.zip

rhamnett, 2020-01-30:
@iPsych the mac link above is a 404

Namburger, 2020-01-30:
Hi guys, the fix is in the work, I'll continue all updates

Namburger, 2020-01-30:
Download link is [here](https://dl.google.com/coral/edgetpu_api/edgetpu_runtime_20200128.zip)!

feranick, 2020-01-30:
I am running MacOS 10.15.3, with python 3.7.6 from macports. I installed the binaries as suggested and the TensorFlow Lite Python wheel package:
https://dl.google.com/coral/python/tflite_runtime-2.1.0-cp37-cp37m-macosx_10_14_x86_64.whl
The installation proceeds with no issues.

When I run python and try to import the tflite_runtime module (which works perfectly in linux), I get:
ModuleNotFoundError: No module named 'tflite_runtime'

The module is installed, not sure why it is not recognized. Thoughts?
BTW: I am aware the package is only available for MacOS 10.14 (and I am running 10.15).

Namburger, 2020-01-30:
@feranick 
I'm not sure how python pip packages handle the case where it's mismatching OS version. However, if you said it installed fine then it should be okay. 
I would definitely check whether is it's installed in the correct python path or not. For instance, my python path can be found this way:
```
$ python3 -c 'print(__import__("sys").path)'    
[...,'~/.local/lib/python3.7/site-packages',...]
```
Which returns a list of paths that python modules are supposed to be installed and then when I look for the path of tflite_runtime, the path should match at least one of the paths in the list above + the `tflite_runtime` directory:
```
$ python3 -c 'print(__import__("tflite_runtime").__path__)'                                             
['~/.local/lib/python3.7/site-packages/tflite_runtime']
```
Unfortunately you'll have to where the `tflite_runtime` module is actually installed in your system and append that to your python path. This should be the same for all python modules so you can reference this if you ever run into this issue again. Hope this helps!

feranick, 2020-01-30:
Thanks @Namburger ! There was a conflict with pip so I had two installations directories for site-packages... All is solved, I am testing the library now. Sorry for the trouble.

feranick, 2020-01-30:
@Namburger  All seems to be working perfectly, thanks very much!

closed 2020-02-02 by chertov