google-coral/edgetpu-platforms #21: master: fails with  imx-vpuwrap-dev deps

powderluv, 2019-10-04:
For a variety of reasons we need to be building mendel from source and using the "master" / day branch based off debian 10.x instead of the current "chef" release. 

Looks like it depends on some internal packages. Could we please remove those deps or add stubs if they can't be removed. 

Thanks

====================================
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 builddeps:/build/imx-gst1.0-plugin_4.4.5-5.dsc:arm64 : Depends: imx-vpu-hantro-dev:arm64 (>= 1.11.0) but it is not going to be installed                                                                                   
                                                        Depends: imx-vpuwrap-dev:arm64 (>= 4.4.5) but it is not going to be installed                                                                                       
E: Unable to correct problems, you have held broken packages.
E: pbuilder-satisfydepends failed.

Namburger, 2019-10-15:
@powderluv 
Are you trying the build by following this guide here?
https://coral.googlesource.com/docs/+/refs/heads/master/GettingStarted.md
Make sure that you are connected to an unrestricted network as the build will requires pulling from debian source. `FETCH_PACKAGES=true m` should download all dependencies.

powderluv, 2019-10-18:
that did the trick.

closed 2019-10-18 by powderluv