google-coral/edgetpu-platforms #27: ModuleNotFoundError: No module named 'edgetpu'

LifelongLearnerOki, 2019-12-29:
Currently, I'm working on a PiCar-V with Respberry 3+, but I have the same issue with "No module named 'edgetpu'.

I have installed the TPU drivers and API’s as follows:
```
wget https://dl.google.com/coral/edgetpu_api/edgetpu_api_latest.tar.gz -O edgetpu_api.tar.gz --trust-server-names
tar xzf edgetpu_api.tar.gz
cd edgetpu_api
bash ./install.sh
```

And when I try to run the live detection test with 
`python3 code/coco_object_detection.py`
(described in this article https://towardsdatascience.com/deeppicar-part-3-d648b76fc0be?)

I always get the following error:
`Traceback (most recent call last): File "code/coco_object_detection.py", line 9, in <module> import edgetpu.detection.engine ModuleNotFoundError: No module named 'edgetpu'`

Do you have any ideas about how to solve this issue?

Thank you.
Cheers

LifelongLearnerOki, 2019-12-29:
Solved by Installing the Edge TPU library as follows

```
sudo apt-get update
sudo apt-get install python3-edgetpu
```

Link: https://coral.ai/docs/edgetpu/api-intro

closed 2019-12-29 by LifelongLearnerOki

Karsten12345, 2020-01-03:
I have the same problem. If I use `sudo apt-get install python3-edgetpu` I get the message " E: Package python3-edgetpu cannot be found"

Many thanks in advance

jingw222, 2020-02-23:
Hey, 
Installing the Debian package `python3-edgetpu` works fine but only under global system environment. However how can I import and use `edgetpu` in my virtual environment? Is there a pypi release available for the package?

Namburger, 2020-02-24:
@Karsten12345 did you add the source list rule and do an apt update?
@jingw222 please refer to this answer: https://github.com/google-coral/edgetpu/issues/1#issuecomment-590433931

PrometheusAlatheia, 2022-02-22:
I am having the same issue. 
#########NO Module named edgetpu############
pi@raspy:~/DeepPiCar/models/object_detection $ python3 code/coco_object_detection.py
Traceback (most recent call last):
  File "/home/pi/DeepPiCar/models/object_detection/code/coco_object_detection.py", line 9, in <module>
    import edgetpu.detection.engine
ModuleNotFoundError: No module named 'edgetpu'
####### Tried install python3-edgetpu#####################
pi@raspy:~ $ sudo apt-get install python3-edgetpu
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 python3-edgetpu : Depends: python3 (< 3.9) but 3.9.2-3 is to be installed
E: Unable to correct problems, you have held broken packages.

hjonnala, 2022-02-22:
@PrometheusAlatheia edgeptu api is deprecated. Please uninstall it and install pycoral from [software center](https://coral.ai/software/#debian-packages).

Please check these sample examples and make the necessary code changes for your script: https://github.com/google-coral/pycoral/tree/master/examples

Thanks!

1joshw, 2023-10-24:
same error

2023-10-24 14:49:05 ERROR:Failed to load the extension "EdgeGPT".
Traceback (most recent call last):
  File "F:\A1111\Chat\text-generation-webui\modules\extensions.py", line 36, in load_extensions
    exec(f"import extensions.{name}.script")
  File "<string>", line 1, in <module>
  File "F:\A1111\Chat\text-generation-webui\extensions\EdgeGPT\script.py", line 8, in <module>
    from EdgeGPT.EdgeGPT import Chatbot, ConversationStyle
ModuleNotFoundError: No module named 'EdgeGPT'