google-coral/edgetpu-platforms #28: AttributeError: module 'enum' has no attribute 'IntFlag'

Luke1962, 2019-12-30:
Hi,
I'm trying to make these libraries working on  Rockpi4 (RK3399 Arm Sbc), with installed on:

- Ubuntu 18.04.03 LTS
- Python versions: 2.7, 3.6 (where edgetpu has been installed) , 3.7
- ROS melodic full desktop (having a lot of modules depending on enum34)
- the following symlink to avoid error rospkg not found

```
cd /usr/local/lib/python3.6/dist-packages/edgetpu/swig/
 sudo ln -s _edgetpu_cpp_wrapper.cpython-35m-aarch64-linux-gnu.so _edgetpu_cpp_wrapper.cpython-36m-aarch64-linux-gnu.so

```

After edgetpu installation, if I run the classic test script:
```
python3 classify_image.py \
--model ~/coral/test/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite \
--label ~/coral/test/inat_bird_labels.txt \
--image ~/coral/test/parrot.jpg
```

  I get the following error:
```
Traceback (most recent call last):
  File "classify_image.py", line 16, in <module>
    import argparse
  File "/usr/lib/python3.6/argparse.py", line 89, in <module>
    import re as _re
  File "/usr/lib/python3.6/re.py", line 142, in <module>
    class RegexFlag(enum.IntFlag):
AttributeError: module 'enum' has no attribute 'IntFlag'
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 53, in apport_excepthook
    if not enabled():
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 24, in enabled
    import re
  File "/usr/lib/python3.6/re.py", line 142, in <module>
    class RegexFlag(enum.IntFlag):
AttributeError: **module 'enum' has no attribute 'IntFlag'**

Original exception was:
Traceback (most recent call last):
  File "classify_image.py", line 16, in <module>
    import argparse
  File "/usr/lib/python3.6/argparse.py", line 89, in <module>
    import re as _re
  File "/usr/lib/python3.6/re.py", line 142, in <module>
    class RegexFlag(enum.IntFlag):
AttributeError: module 'enum' has no attribute 'IntFlag'
```
I've only found on internet, as working solution, to uninstall enum34 with:

Unfortunately, this operation
` sudo apt-get remove python-enum34`
 would uninstall a lot of ros packages I need.
`
rock@rockpi4:~/coral/test$ sudo apt-get remove python-enum34
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  cython default-libmysqlclient-dev fltk1.3-doc fluid freeglut3 freeglut3-dev
  gazebo9 gazebo9-common gazebo9-plugin-base gdal-data google-mock googletest
  hdf5-helpers ibverbs-providers libaec-dev libaec0 libapr1 libapr1-dev
  libaprutil1 libaprutil1-dev libarmadillo-dev libarmadillo8 libarpack2
  libarpack2-dev libassimp-dev libassimp4 libassuan-dev libavdevice-dev
  libavfilter-dev libavresample-dev libblas-dev libboost-all-dev
  libboost-atomic-dev libboost-atomic1.65-dev libboost-atomic1.65.1
[... 203 lines cut ...]
  ros-melodic-rqt-launch ros-melodic-rqt-moveit ros-melodic-rqt-msg
  ros-melodic-rqt-nav-view ros-melodic-rqt-plot ros-melodic-rqt-pose-view
  ros-melodic-rqt-publisher ros-melodic-rqt-py-common
  ros-melodic-rqt-reconfigure ros-melodic-rqt-robot-dashboard
  ros-melodic-rqt-robot-monitor ros-melodic-rqt-robot-plugins
  ros-melodic-rqt-rviz ros-melodic-rqt-service-caller ros-melodic-rqt-srv
  ros-melodic-rqt-tf-tree ros-melodic-rviz ros-melodic-rviz-plugin-tutorials
  ros-melodic-rviz-python-tutorial ros-melodic-simulators
  ros-melodic-smach-ros ros-melodic-stage-ros ros-melodic-tf
  ros-melodic-tf-conversions ros-melodic-tf2-geometry-msgs ros-melodic-tf2-kdl
  ros-melodic-tf2-ros ros-melodic-turtle-actionlib ros-melodic-turtle-tf
  ros-melodic-turtle-tf2 ros-melodic-urdf-sim-tutorial
  ros-melodic-urdf-tutorial ros-melodic-visualization-tutorials
  ros-melodic-viz ros-melodic-xacro
0 upgraded, 0 newly installed, 96 to remove and 0 not upgraded.
After this operation, 115 MB disk space will be freed.
Do you want to continue? [Y/n] 

`
I've also installed 
`pip install enum-compat
`
but seems not working (same error).

Is there a solution to point explicitly to enum standard library without removing enum34?

sohartma, 2020-04-30:
I had the same issue. Wanted to use the system python (2.7), but cmake used python3.6 somehow. I edited the python path in the catkin_find_pkg file, which equals to your argparse.py file.
So with python 2 it worked, no error. 
I guess, check that your python paths are correct. Or go with python2.

manoj7410, 2021-07-01:
Feel free to reopen if this is still an issue.

closed 2021-07-01 by manoj7410

nagarjunvinukonda, 2023-08-02:
I have the same issue:

Any solution?