google-coral/edgetpu-platforms #31: all examples fail on Raspberry Pi Zero with Buster Lite

cipy, 2020-06-10:
with

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/edgetpu/swig/edgetpu_cpp_wrapper.py", line 18, in swig_import_helper
    return importlib.import_module(mname)
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 965, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'edgetpu.swig._edgetpu_cpp_wrapper'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "classify_image.py", line 19, in <module>
    from edgetpu.classification.engine import ClassificationEngine
  File "/usr/local/lib/python3.7/dist-packages/edgetpu/classification/engine.py", line 17, in <module>
    from edgetpu.basic.basic_engine import BasicEngine
  File "/usr/local/lib/python3.7/dist-packages/edgetpu/basic/basic_engine.py", line 15, in <module>
    from edgetpu.swig.edgetpu_cpp_wrapper import BasicEngine
  File "/usr/local/lib/python3.7/dist-packages/edgetpu/swig/edgetpu_cpp_wrapper.py", line 21, in <module>
    _edgetpu_cpp_wrapper = swig_import_helper()
  File "/usr/local/lib/python3.7/dist-packages/edgetpu/swig/edgetpu_cpp_wrapper.py", line 20, in swig_import_helper
    return importlib.import_module('_edgetpu_cpp_wrapper')
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ModuleNotFoundError: No module named '_edgetpu_cpp_wrapper'
```

this is Buster Lite on RPi Zero W

The examples do work on rpi0-raspbian-stretch-edgetpu-1.9.2.img.gz (ancient OS image)

zamathebest, 2020-08-20:
Hi, same issue with the RPi0-W using Raspbian Buster
May you share the makefile and Dockerfile used to compile the edgetpu for RPi0?

hjonnala, 2021-11-22:
Closing   the  issue since the edgetpu is deprecated and due to lack of official support for Raspberry Pi Zero. Sorry for the inconvenience.

closed 2021-11-22 by hjonnala